adafruit/Adafruit_CircuitPython_LSM6DS #5: Add shake?

caternuson, 2020-02-25:
The CircuitPlayground library just did this by reusing the LIS3DH implementation. So maybe this is something that should be added to the LSM6DS library? But starting the conversation here. Since this feature is just math on the x,y,z accelo readings, it's not really hardware specific to the LSM6DS.

ladyada, 2020-03-02:
there's 'significant motion' detection, wanna try adding?

caternuson, 2020-03-02:
Hmm, wonder if it would work for shake? Might be more of a heavily integrated accelo reading to determine change in position:
[image]
Can at least check behavior first.

ladyada, 2020-03-02:
🤷‍♀ want me to write some arduinocode?

caternuson, 2020-03-02:
Sure, if you want, or I can. I like the way they put it in quotes. So just some simple sketch to test what actual behavior is.

ladyada, 2020-03-03:
tried it. its not that great - basically same as pedometer, you have to shake for at  least a second to get it to fire.

ladyada, 2020-03-03:
tried wakeup instead - seems better. wanna try the arduino side?

caternuson, 2020-03-03:
just wanna check in case it matters - which LSM6DS are you testing with? it looks like these basic detection features are common, but i might be missing something?

ladyada, 2020-03-03:
im using the LSM6DS33 - other subclass chips inherit all the same functionalities

caternuson, 2020-03-03:
ok, for arduino side, added some info here:
https://github.com/adafruit/Adafruit_LSM6DS/issues/9

godzan, 2020-04-25:
Is there any way to get this working with circuitpython?

caternuson, 2020-04-30:
@godzan It will eventually. It's being worked out on the Arduino side first. You commented there as well.