adafruit/Adafruit_CircuitPython_LSM6DS #9: gyro values return in degrees per second but CP standard is radians per second

kevinjwalters, 2020-03-07:
Based on a quick test with an Adafruit CLUE using the https://github.com/adafruit/Adafruit_CircuitPython_CLUE library which is passthru for gyro value, the gyro return appears to be in degrees per second. The code in the example (and README) also suggest this is the case in the textual description of units: https://github.com/adafruit/Adafruit_CircuitPython_LSM6DS/blob/master/examples/lsm6ds33_simpletest.py#L12

https://circuitpython.readthedocs.io/en/latest/docs/design_guide.html#sensor-properties-and-units specifies `gyro` as `x, y, z radians per second` (NB, inconsistent with `orientation` specification).

ladyada, 2020-03-07:
could be! @siddacious ya can take a look next week

kevinjwalters, 2020-03-09:
BTW, you can dangle a board by its USB lead and twist that to get approximately 60 rpm as a crude measurement test if you don't have a spinnable platform to hand.

kevinjwalters, 2020-03-30:
Similar issue over on https://github.com/adafruit/Adafruit_BNO055/issues/50 but perhaps that can't be changed because too much code + users would be affected.

siddacious, 2020-04-16:
sorry for the ate reply but yes, I can confirm that we're reporting dps not rads. I'll put in a fix and do a major release since the API is changing despite it also being a fix

commit 350d313d (2020-04-16): BREAKING CHANGE - change from dps->rads for gyro to fix #9
commit 24224cc9 (2020-04-20): Merge pull request #14 from adafruit/rads_fix

change from dps->rads for gyro to fix #9
closed 2020-04-20 by jepler