adafruit/Adafruit_CircuitPython_LSM6DS #15: Memory Allocation issue on M0, others?

caternuson, 2020-05-01:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=19&t=164655

And my testing:
```
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit Metro M0 Express with samd21g18
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 4, in <module>
  File "adafruit_lsm6ds.py", line 188, in <module>
  File "adafruit_lsm6ds.py", line 217, in LSM6DS
MemoryError: memory allocation failed, allocating 184 bytes
```

ladyada, 2020-05-01:
yeah its a large library - you'll need an M4/nRF for this.

that said, there are a few registers that are not used 
https://github.com/adafruit/Adafruit_CircuitPython_LSM6DS/blob/master/adafruit_lsm6ds.py#L229
@siddacious please comment out unused registers

tannewt, 2020-05-01:
Are new libraries tested on M0s? Shouldn't we target the smallest platform for them? Making them small means they'll leave more room for user code on larger platforms as well.

It could be worth adding a memory size check to the library CI so it's prevalent how big each is.

ladyada, 2020-05-01:
this one was only tested on nrf52 (cause its used in the feather sense & clue) - a memory size check would be nice but not sure how we could calculate that?

tannewt, 2020-05-04:
mpy size could be an approximation. I bet we could have mpy-cross actually output it since it loads everything into memory before writing it out to an mpy. Not a simple thing to do though.

siddacious, 2020-05-05:
I removed the unused registers but it still doesn't fit on the CPX.  @ladyada lmk what if anything else you'd like me to do here

ladyada, 2020-05-05:
nope thats a good place to take a break, it may just not fit

siddacious, 2020-06-06:
@ladyada I'm un-assigning myself for now until there's something tangible for me to do

ladyada, 2020-06-06:
kk

jposada202020, 2021-05-19:
Just wondering what we would like to do with this issue,I do not have a metro express but this test ok in a QT Py.

```python
Adafruit CircuitPython 6.1.0 on 2021-01-21; Adafruit QT Py M0 with samd21e18
>>> import lsm6ds_lsm6ds33_simpletest
Acceleration: X:0.80, Y: 0.83, Z: 9.97 m/s^2
Gyro X:0.05, Y: -0.09, Z: -0.04 radians/s

Acceleration: X:0.81, Y: 0.84, Z: 9.96 m/s^2
Gyro X:0.05, Y: -0.09, Z: -0.04 radians/s

Acceleration: X:0.81, Y: 0.83, Z: 9.96 m/s^2
Gyro X:0.05, Y: -0.09, Z: -0.04 radians/s
```

caternuson, 2022-03-17:
Still hitting the issue on Metro M0 Express:

```python
Adafruit CircuitPython 7.2.3 on 2022-03-16; Adafruit Metro M0 Express with samd21g18
>>> import lsm6ds_lsm6ds33_simpletest
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "lsm6ds_lsm6ds33_simpletest.py", line 6, in <module>
  File "adafruit_lsm6ds/__init__.py", line 158, in <module>
  File "adafruit_lsm6ds/__init__.py", line 190, in LSM6DS
MemoryError: memory allocation failed, allocating 184 bytes
>>> 
```

ladyada, 2022-03-19:
i think we just gotta update the readme and guide to say "SAMD21 can't use this advanced IMU"