adafruit/Adafruit_CircuitPython_LSM6DS #17: SPI support?

pslade2, 2020-06-01:
I saw that there is an Arduino SPI library for this device. Are there plans to support SPI for this sensor in python as well?

ladyada, 2020-06-01:
we haven't gotten to it, but if you'd like to help we'd love a hand :)

ladyada, 2020-10-01:
@sakethramanujam you would need a sensor, do you have one?

sakethramanujam, 2020-10-01:
I'd need a sensor. Where can I procure one from?

ladyada, 2020-10-01:
we sell them at adafruit, but you could get them anywhere as long as its an LSM6DS series

sakethramanujam, 2020-10-02:
Trying to find one at my local store.

jposada202020, 2021-04-24:
@pslade2 hello, any luck getting the sensor? thanks. :)

rjsdotorg, 2025-03-16:
I have a 4517 and 4565, currently wired for I2C and running.
Pi zero 2 W BTW.
Once I work out some other things I will need to go to SPI, as the bandwidth for 9 channels + temp at max rates eat up I2C.
I went from Pico to Pi Zero2 W as I found that the Pico is struggle to just sample and save data at high rates. I checked out the code at Github but as I posted here https://forums.adafruit.com/viewtopic.php?p=1049398#p1049398 " I'm not following (yet) how self.i2c_device is even used as it does not seem to be referenced"

rjsdotorg, 2025-03-16:
There is also an issue mentioned elsewhere of getting (near) simultaneous data from the two chips. It could use a helper function to grab all board data in one go, and document corresponding DRDY wire connection. At St there is some posting on this, like https://community.st.com/t5/mems-sensors/lsm6dsl-int2-lt-gt-lis3mdl-drdy-short-circuit/m-p/775622

Related to this, I2C "Fast Mode" appears to be 400kHz on the Pi driver, but the ST part supports "I²C fast mode +" of 1MHz, and this https://www.abelectronics.co.uk/kb/article/1/i2c-part-2-enabling-i2c-on-the-raspberry-pi, and the Camera Module docs, implies that the Zeros might be able to run at 1MHz. I'll try to test.

rjsdotorg, 2025-03-17:
FYI, I just added `frequency=1000000` to the busio.I2C call, and I now can call for all 9 values in a tight loop ~1440/s while storing in a ulab array.
This compares to ~850/s with` frequency=400_000`, so the Pi Zero 2 W is doing I2C faster.

BTW the Pi Zero 2 W does show as "raspberrypi_zero2w" in `board.board_id`, but `adafruit_platformdetect.Detector()` returns None for id and board.