adafruit/Adafruit_CircuitPython_LSM6DS #18: User report: accelerometer and gyro didn't work without explicit data rate setting

jepler, 2020-06-07:
Discord user "Pamz" reported on #circuitpython:
> Hi, I am testing Bluefruit feather sense I received couple days ago using this demo code: https://learn.adafruit.com/adafruit-feather-sense/circuitpython-sense-demo 
I am watching response in serial. Everything seems to be working, except gyro and acceleration. I get 0 values no mather how much I move it.
default lsm6ds33 accelerometer_data_rate and gyro_data_rate is at 0Hz. If I set it to 1066Hz, it works like a charm, otherwise it is just zeros.

I looked at the code real quick and it looks like it _should_ be setting RATE_104_HZ at startup, so I'll admit to being confused.

I'm not sure if this is related to a comment internally on Slack that there was "some variant of the LSM6DS33 that completely dies when you tell it to reboot", or if that recent mention is just a coincidence.

jepler, 2020-06-07:
@siddacious maybe you can clarify whether it could be related to the reset issue or whether that's likely a red herring?

siddacious, 2020-06-09:
@jepler I'll take a look at this today. FWIW, the discussion  on slack was regarding sending a _boot_ command to the sensor which clears the memory on the device. It's not required for normal functioning so we removed it:
https://github.com/adafruit/Adafruit_CircuitPython_LSM6DS/commit/2ffd1a03e7dc647ea185118e2605f147429a4eca

I'll give it a test to see if I can reproduce it. And update "Pamz" either way

siddacious, 2020-06-12:
I tested with the latest version of the library and didn't have any issues with my feather sense.

siddacious, 2020-06-19:
I didn't receive a response so I'll re-open this if I receive one that indicates there is still a problem

closed 2020-06-19 by siddacious