adafruit/Adafruit_CircuitPython_LSM6DS #21: AttributeError: 'LSM6DS33' object has no attribute 'pedometer_steps'

dglaude, 2020-07-30:
Both the example lsm6ds_pedometer.py and the learn guide https://learn.adafruit.com/clue-step-counter-st-lsm6ds33/overview do fail on trying to get pedometer_steps.

caternuson, 2020-07-30:
Hmm. Yah, it's not there:
```python
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit CLUE nRF52840 Express with nRF52840
>>> import board
>>> import adafruit_lsm6ds
>>> adafruit_lsm6ds.__version__
'4.0.1'
>>> sensor = adafruit_lsm6ds.LSM6DS33(board.I2C())
>>> dir(sensor)
['__class__', '__dict__', '__init__', '__module__', '__qualname__', 'reset', 'i2c_device', '_chip_id', '_raw_accel_data', '_raw_gyro_data', '_accel_range', '_accel_data_rate', '_gyro_data_rate', '_gyro_range', '_gyro_range_125dps', '_gyro_range_4000dps', '_sw_reset', '_bdu', '_high_pass_filter', '_i3c_disable', '_pedometer_reset', '_func_enable', '_ped_enable', 'CHIP_ID', 'acceleration', 'gyro', '_scale_xl_data', '_scale_gyro_data', 'accelerometer_range', 'gyro_range', 'accelerometer_data_rate', 'gyro_data_rate', 'pedometer_enable', 'high_pass_filter', '_cached_accel_range', '_cached_gyro_range']
>>> 'pedometer_enable' in dir(sensor)
True
>>> 'pedometer_steps' in dir(sensor)
False
>>>  
```

caternuson, 2020-07-30:
Looks like it got removed in this PR:
https://github.com/adafruit/Adafruit_CircuitPython_LSM6DS/pull/16
Most likely a mistake. The other pedometer related registers were refactored. That one got removed.

dherrada, 2020-08-06:
Fixed by #20

closed 2020-08-10 by tannewt