adafruit/Adafruit_CircuitPython_LSM6DS #31: Change LSM6DS33 address

giladgar, 2020-12-03:
Hi, 
While trying to run the example code on my Raspberry Pi 4 [model B]
`# SPDX-FileCopyrightText: Copyright (c) 2020 Bryan Siepert for Adafruit Industries
#
# SPDX-License-Identifier: MIT
import time
import board
import busio
from adafruit_lsm6ds.lsm6ds33 import LSM6DS33
 
i2c = busio.I2C(board.SCL, board.SDA)
 
sensor = LSM6DS33(i2c)
 
while True:
    print("Acceleration: X:%.2f, Y: %.2f, Z: %.2f m/s^2" % (sensor.acceleration))
    print("Gyro X:%.2f, Y: %.2f, Z: %.2f radians/s" % (sensor.gyro))
    print("")
    time.sleep(0.5)` 

I got an `ValueError : No I2C debice at address: 6a`

I know there is a way to change the I2C address to 6b with the SD0 pin, but i wanted to free this pin for future expansion [maybe connecting two LSM6DS33's to the same Pi]. 
Is there a way to tell my Pi to read from this specific address?

Thank you (:

giladgar, 2020-12-03:
From the library documentation i found the syntax for

ladyada, 2020-12-07:
you can pass it into init 
`sensor = LSM6DS33(i2c, address=0x6b)`
https://github.com/adafruit/Adafruit_CircuitPython_LSM6DS/blob/master/adafruit_lsm6ds/__init__.py#L199

closed 2020-12-07 by ladyada