adafruit/Adafruit_CircuitPython_LSM6DS #32: Missing data read flag

David-Mason-1992, 2021-01-02:
Hi,

I've just started using the library and it was really easy to get up and running so good job there!

For my application (datalogger) it is important i read every single reading from the accel exactly once. It is also important if for some reason i miss a reading that I can tell if that has occured.

I think the easiest way to implement this would be to add a flag (or two? one for gyro, one for accel) to the class variables which can be checked before calling `sensor.acceleration` or `sensor.gyro`. Upon calling those functions the function would set the flag to 0 to indicate no values have changed since last read. 

Within the function that gets the data over I2C or handles the update of that data it should update the flag each time adding +1 to it's value. This would mean if say 20 readings were missed the user could easily tell by looking at the flag values.

I considered trying to implement this by say checking values were consistant but this is not reliable as during static portions of my testing I will likely get the same values for a few readings.

I am happy to try help implement this if you think this is a viable route to go but as I am not the most proficient Python programmer I can't for the life of me work out where the function is that gets the raw values over I2C to increment the flag within?

ladyada, 2021-01-03:
i dont think we would take a counter addition, it isnt something that really makes sense in the scheme of the library - usually if you miss data points you know because time passed (since the accel is going at a fixed rate and you are timestamping with time.monotonic()), or you can set up the FIFO which is the 'proper' solution

if you want to add FIFO support we'd totally welcome a PR!

David-Mason-1992, 2021-01-03:
I read about the FIFO in the data sheet. It was what I was hoping to use but obviously saw it wasn't implemented.

Given that i can't work out where the I2C communication was happening or how the frequency that the communication is happening is controlled i don't think I have the ability to impement that. Will take a look but not overly optimistic hah!

ladyada, 2021-01-04:
the i2c system is kinda nifty it works with the `register` library so you just need to make the registers for the chip needed to enable the FIFO and then write the data ya need https://github.com/adafruit/Adafruit_CircuitPython_LSM6DS/blob/master/adafruit_lsm6ds/__init__.py#L175