adafruit/Adafruit_CircuitPython_LSM6DS #35: Error while running code from the examples

TinManAkshay, 2021-02-08:
Hi,

I am trying to run lsm6dsox.py and but I am getting this error. 

pi@raspberrypi:~/Adafruit_CircuitPython_LSM6DS/adafruit_lsm6ds $ sudo python3 lsm6dsr.py 
Traceback (most recent call last):
  File "lsm6dsr.py", line 7, in <module>
    from . import LSM6DS, LSM6DS_DEFAULT_ADDRESS, LSM6DS_CHIP_ID
SystemError: Parent module '' not loaded, cannot perform relative import

What does it mean by parent module here? Do i have to run sudo pip3 install Adafruit_CircuitPython_LSM6DS before I run any of the example codes?

caternuson, 2021-02-08:
Try running the example from this library:
https://github.com/adafruit/Adafruit_CircuitPython_LSM6DS/blob/master/examples/lsm6ds_lsm6dsox_simpletest.py
without sudo.

TinManAkshay, 2021-02-08:
When I click on the above link, it comes back to the original library. I don't get it. Isn't it same?

caternuson, 2021-02-08:
What is `lsm6dsr.py` in your example?
```
pi@raspberrypi:~/Adafruit_CircuitPython_LSM6DS/adafruit_lsm6ds $ sudo python3 lsm6dsr.py
Traceback (most recent call last):
File "lsm6dsr.py", line 7, in
from . import LSM6DS, LSM6DS_DEFAULT_ADDRESS, LSM6DS_CHIP_ID
SystemError: Parent module '' not loaded, cannot perform relative import
```

TinManAkshay, 2021-02-08:
Actually I have developed that file to support LSM6DSR module and using this library i have been trying to communicate with that IMU. 

I2C address required is 0x6B
Chip ID is 0x6B.

These changes i have made in __init__.py.

caternuson, 2021-02-08:
This library does not currently support the LSM6DSR.

TinManAkshay, 2021-02-08:
That I know it doesn't. But if I want to support LSM6DSR, what changes I have to make in the library other than i2c address and chip ID. Can you please give me any info. on that? 

Thanks,
Akshay

caternuson, 2021-02-08:
You would need to read through the LSM6DSR datasheet and adapt whatever is needed for its specific register layout. The I2C address only deals with talking over I2C. The specific I2C traffic will be specific to the sensor - which is where the datasheet comes in.

Any of the other sensor specific modules is the library could serve as good examples: 
https://github.com/adafruit/Adafruit_CircuitPython_LSM6DS/tree/master/adafruit_lsm6ds

ladyada, 2021-02-08:
closing since we dont support the chip in question

closed 2021-02-08 by ladyada