adafruit/Adafruit_CircuitPython_LSM6DS #42: ImportError: Cannot import name

JonahMorgan, 2021-05-26:
I'm trying to import the library on CircuitPython for the LSM6DSOX, but it keeps throwing an error on this line: 
```
>>> from adafruit_lsm6ds.lsm6dsox import LSM6DSOX
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name LSM6DSOX
```
The first time I ran it, it gave me a different error: 
`'ImportError: no module named 'adafruit_lsm6ds.LSM6DS'`
but now it throws the first one every time. 

This is running on the Arduino Nano RP2040 Connect using CircuitPython 7.0.0. I'm using the Mu REPL to communicate. I have the libraries loaded to the lib/ folder in .MPY format. 

The same behavior happens when trying to import from any of the LSM6DS modules. The problem is not with the REPL, because I can blink the led just fine, and I can also use other libraries. 

This is my first issue, so I apologize if the formatting is off.

JonahMorgan, 2021-05-26:
I just redownloaded the library and reinstalled it and now everything works as expected. 
Closing the issue now.

closed 2021-05-26 by JonahMorgan

TinManAkshay, 2021-06-07:
Hi @JonahMorgan,

Can you please share how did you resolve the issue you were getting? 

I have the same LSM6DSOX with me and have installed the libraries system-wide (using sudo pip3). I am able to get the example working on the terminal and observed the correct behavior of the IMU. However, my main .py file is in /usr/bin/xxx.py and I am importing the same libraries into that file. When i run the at start-up, it does not recognize LSM6DSOX libraries and I believe I might be getting the same error as you were getting.

I'd appreciate if you can share some suggestions or solutions to it?

Thanks,
Akshay

JonahMorgan, 2021-06-12:
Hey @TinManAkshay, 

If you are using CircuitPython, it's important to use the CircuitPython library bundles and install them onto the CircuitPython drive, as installing system-wide won't work. For the RP2040 Nano Connect, double clicking the reset button enters it into bootloader mode. At that point, drag and drop the uf2 flash file for CPy to the RPI drive that should appear. The Connect automatically restarts and enters CircuitPython mode. From here, drag the LSM6DS library folder from the Adafruit CircuitPython compiled library (that should be found in the releases of this repository) and paste it into the /lib/ folder of the CircuitPython drive. then place your main.py (code.py for me, but it shouldn't matter) into the root directory of the drive. The code is automatically detected and run by the bootloader, so you should see results immediately. 

I hope this solves your problem. 

Hope I could help, 
Jonah