adafruit/Adafruit_CircuitPython_LSM6DS #48: Wake on Significant Motion

scath999, 2022-02-16:
Wondering how I can configure this device to fire an interrupt on significant motion, per page 45 of the datasheet ([here](https://www.mouser.ca/datasheet/2/389/dm00157718-1798631.pdf))?

The goal is to be able to set an alarm in CPy to wake when pin D3 is set high - according to the schematics for the Feather Sense (which I'm developing on), this is the GPIO connected with INT1 on the LSM6DS33.

Something like

``` 
pin_alarm = alarm.pin.PinAlarm(pin=board.D3, value=True, pull=True)
alarm.light_sleep_until_alarms(pin_alarm)
```

The datasheet seems to indicate that setting register `0x0D` on the device to 64 (bin 01000000) will accomplish this.

Thoughts?

tekktrik, 2022-03-15:
Looks like this should be an easy addition.  I don't have the hardware but I can submit a PR with this functionality if you'd like to help me test it!

tekktrik, 2022-03-15:
I plan to get one anyways to work on some other issues, so no worries!

caternuson, 2022-03-15:
Seems related to:
https://github.com/adafruit/Adafruit_CircuitPython_LSM6DS/issues/5
which I appear to have totally forgotten about and abandoned. :(

@tekktrik I can try picking this up. There's a similar outstanding issue on the Arduino side as well.

tekktrik, 2022-03-16:
Sounds good!  I was going to see about doing #17 so I can just try that when the sensor arrives!

caternuson, 2022-03-16:
This library supports multiple devices with a common base class `LSM6DS` defined in `__init__.py`. There is some variability in the enabling and use of Significant Motion Detection between the devices. Here's a summary taken from several app notes.

<hr/>

**LSM6DS33 (?? guessing ??)** STM doesn't seem to acknowledge its existence anymore
**LSM6DS3TR-C (AN5130)**

[image]

<hr/>

**ISM330DHCX (AN5398)**
**LSM6DSOX (AN5272)**
**LSMM6DSO32 (AN5473)**

[image]

caternuson, 2022-03-16:
@scath999 In AN5398, it describes the "significant motion" detection as:

> It generates an interrupt when the difference between the number of steps counted from its initialization/
> reset is higher than 10 steps.

Is that what you are expecting? Or were just wanting something like shake detection?

scath999, 2022-03-17:
Hi Everyone,

Sorry for the late replies - busy at work.

I've managed to solve this issue by inspecting the schematics for the Feather. It shows that this device's IRQ1 is connected to D3. 

I put a Pin.Alarm on that pin and include it in the list of alarms to wake from.

Thank you so much for all the input, and sorry for not updating.

Should I close this ticket?

scath999, 2022-03-17:
> @scath999 In AN5398, it describes the "significant motion" detection as:
> 
> > It generates an interrupt when the difference between the number of steps counted from its initialization/
> > reset is higher than 10 steps.
> 
> Is that what you are expecting? Or were just wanting something like shake detection?
> 
> **EDIT**: Also checkout the comment here: [#5 (comment)](https://github.com/adafruit/Adafruit_CircuitPython_LSM6DS/issues/5#issuecomment-593989865) It may not be what you're expecting it to be?

Exactly - @ladyada is having the same experience I'm having.