adafruit/Adafruit_CircuitPython_LSM6DS #51: implement Machine Learning for LSM6DSOX?

jerryneedell, 2022-03-05:
The MicroPython driver for the LSM6DOX has example for using ST's Machine Learning Core configurations  for detecting certain types of motions.

https://github.com/micropython/micropython/blob/master/drivers/lsm6dsox/lsm6dsox_mlc.py
https://github.com/STMicroelectronics/STMems_Machine_Learning_Core/tree/master/application_examples/lsm6dsox

I have tried the vibration and head gesture examples and they are seem to work.
Would there be any interest in adding the ability to load the "ucf" files via the CircuitPython driver.
We won't be able to use the Interrupt MODE, but could poll.

The change to the library would be to add the "load_mlc, read_mlc_output and set_embedded_functions" methods. There actual usage is done in the examples.  

Let me know if you think this is worth adding for the lsm6dsox.  It may also apply to the lsmds33 but does not appear to be supported on the lsmdso32.

ladyada, 2022-03-05:
eeeeenteresting. i dont think it works on anything that doesnt end in an 'x' - but yeah go ahead! i dont think we can commit the ucf files right?

jerryneedell, 2022-03-05:
not sure about committing the .ucf's  I guess we can just provide a link to them, That is what MP did.
I tried a few more demos -- door opening and position detection -- worked well with MP.

I'll go ahead and see if I can get it added to the CP library then we can figure out how to present it.

jerryneedell, 2022-03-20:
implemented by #52

closed 2022-03-20 by jerryneedell